xmos/sln_voice #129: Implement software restructuring so xcore_sdk is not a dependency

keithm-xmos, 2022-11-11:


xmos-jmccarthy, 2023-01-19:
All targets renamed.

fwk_core complete
https://github.com/xmos/fwk_core/pull/15

fwk_io pending on inferencing branch to allow tests to pass.
https://github.com/xmos/fwk_io/pull/48

fwk_rtos complete.
https://github.com/xmos/fwk_rtos/pull/92

sln_voice complete (docs not working currently)
https://github.com/xmos/sln_voice/pull/151

xcore_sdk pending official new name.  repo name, and some internal target names need to be updated to this new name.  My current proposed name is sln_explorer_board
https://github.com/xmos/xcore_sdk/pull/576

xmos-jmccarthy, 2023-01-20:
lib_tflite_micro handled a different way per our discussion.  Removed from fwk_core, and fetch_content enabled in sln_voice directly.

keithm-xmos, 2023-01-21:
Create issue for renaming xcore_sdk https://github.com/xmos/xcore_sdk/issues/577

Closing this issue as all other tasks are ticked off.

closed 2023-01-21 by keithm-xmos