xmos/sln_voice #168: Exception running example_ffva_ua.xe during SRC test

keithm-xmos, 2023-01-30:
Excerpt from Jenkins log:

```
[2023-01-27T19:35:20.738Z] xrun: Program received signal ET_ECALL, Application exception. 
[2023-01-27T19:35:20.738Z]       [Switching to tile[0] core[2]] 
[2023-01-27T19:35:20.738Z]       ep_transfer_complete (ctx=0xc31d8, len=0x0, is_setup=0x0) at /sln_voice/modules/rtos/modules/drivers/usb/src/rtos_usb.c:91 
[2023-01-27T19:35:20.738Z] 
[2023-01-27T19:35:20.738Z]       91    /sln_voice/modules/rtos/modules/drivers/usb/src/rtos_usb.c: No such file or directory.
[2023-01-27T19:35:20.738Z]           in /sln_voice/modules/rtos/modules/drivers/usb/src/rtos_usb.c
```

Full log is here: http://srv-bri-jcim0:8080/blue/organizations/jenkins/XMOS%2Fsln_voice/detail/PR-164/8/pipeline#step-254-log-105

keithm-xmos, 2023-02-03:
Happened again: 
http://srv-bri-jcim0:8080/blue/organizations/jenkins/XMOS%2Fsln_voice/detail/PR-180/2/pipeline#step-874-log-101

xmos-jmccarthy, 2023-02-06:
Still could not reproduce locally.

It appears that when this issue happens the host send data before the device was expecting it since the interface was never set.  Need to take a closer look into cleanup on the host side and may need to consider getting an xtagctrl setup.

keithm-xmos, 2023-02-09:
Seems to be resolved.  Closing but will continue to monitor.

xmos-jmccarthy, 2023-02-14:
Resolved with root cause found.  See #199

PR #199 (2023-02-14): Ensure app tx's data when required
Closes #168 by fixing the issue described below.

The bug seen in #168 appears to be a Linux usb driver behavior (which is why this has not been seen on macOS) that is caused by our application code not sending a smaller than `tx_size_bytes_rate_adjusted` usb packet.  This incorrect behavior was a carry over from the previous synchronous code.  It resulted in the host driver keeping a session open.  This would cause the next usb audio test to run to fail, since the host would send a packet to an endpoint that was not yet configured. 

Latest PR that hit this  [... cut]
closed 2023-02-14 by xmos-jmccarthy