xmos/sln_voice #194: Low Power Race Conditions

xmos-jmccarthy, 2023-02-09:
There are a few possible race conditions in the lp code as of 66eab7cc25a4ec5479f01301804ededea6e2738d.

One case is due to the nature of the variable latency that occurs when using the intertile driver. There is a lot of communication between the two tiles in various threads. There were at least 3 runtime cases seen (not yet reliably reproducible) where tile 0 would be powered down while an intertile transaction was in progress, resulting in a deadlock.  This is potentially caused by the fact that order of processing intertile messages is not guaranteed or by a concurrency bug.

A few cases of concurrency issues can potentially happen.  The experimental branch https://github.com/xmos/sln_voice/tree/experiment/lp_investigation resolves these issues by having a single centralized thread that receives operation requests from the various threads that need to talk to it.  This adds some additional latency, but solves the multiple thread access without littering the code with mutexes or critical sections.  One example of this is the relationship on tile 0 between the [ power control task ](https://github.com/xmos/sln_voice/blob/66eab7cc25a4ec5479f01301804ededea6e2738d/examples/ffd/src/power/power_control.c#L307) and [ low_power_handler ](https://github.com/xmos/sln_voice/blob/66eab7cc25a4ec5479f01301804ededea6e2738d/examples/ffd/inference/wanson/wanson_inf_eng.c#L169).  If tile 1's full power timeout has expired, `inference_engine_low_power_accept()` will be called, which wants to set `inference_power_state`.  There is the potential for the `wanson_engine_task` to be running and in the `timeout_event_handler()` or `low_power_handler()` which both may set `inference_power_state` resulting in potentially undefined behavior due to a state transition being missed or corrupted.  The experimental code performs any state transition within a single ["operation" ]( https://github.com/xmos/sln_voice/blob/27fd6bc84284bbd522a188ac39334a011bcbb1a7/examples/ffd/src/power/lp_control.c#L127) ensuring that atomic state transitions occur.

Another issue occurs around the power state, where the `audio_pipeline_output()` task calls [`power_state_data_add()`](https://github.com/xmos/sln_voice/blob/66eab7cc25a4ec5479f01301804ededea6e2738d/examples/ffd/src/power/power_state.c#L53), which sets `power_state` to POWER_STATE_FULL.  If [power_control_task](https://github.com/xmos/sln_voice/blob/66eab7cc25a4ec5479f01301804ededea6e2738d/examples/ffd/src/power/power_control.c#L307) is in `low_power_ready()` the power_state could be set to an unexpected value between the 2 parallel calls with potentially different inputs to:
https://github.com/xmos/sln_voice/blob/66eab7cc25a4ec5479f01301804ededea6e2738d/examples/ffd/src/power/power_state.c#L30-L39

I suspect that the latter issue is the cause of the lockups I have seen where tile 0 is powered down during an intertile transaction, but at this time I cannot reliably reproduce to debug further.

xmos-jmccarthy, 2023-02-09:
Other noteable.  The issue is very sporadic.  It seems to be more likely to happen when noise is stable and then abruptly loud.  Time to happen has varied from seconds to 30+ minutes.

mbanth, 2023-09-15:
This bug is now tracked in Jira [AP-119](https://xmosjira.atlassian.net/browse/AP-119).

closed 2023-09-15 by mbanth