xmos/sln_voice #217: tag `v1.0.0.0-rc1` did not kick off release actions

keithm-xmos, 2023-02-28:


keithm-xmos, 2023-02-28:
It looks like the `-rc1` is the issue, and we can simply append a * to the pattern to support this and other tag formats or we could have it more specifically support an optional -rc[0-9]+ (not sure if this is the exact regex).

jcarrier-xmos, 2023-02-28:
This pattern has been tested on an experimental branch to support two tag forms:
```
    tags:
      - v[0-9]+.[0-9]+.[0-9]+
      - v[0-9]+.[0-9]+.[0-9]+\-rc[0-9]+
```

Example tags that should trigger with this configuration:
- `v1.0.0`
- `v1.10.10`
- v1.0.0-rc10`

Related docs on syntax:
https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#onpushbranchestagsbranches-ignoretags-ignore
https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#filter-pattern-cheat-sheet

closed 2023-02-28 by keithm-xmos