xmos/sln_voice #232: How-to documents

keithm-xmos, 2023-03-14:
Add How-To documents that could include:

- [ ] How to modify the code for a hybrid USB/I2S architecture. i.e. USB output interface but taking reference signal from I2S
- [ ] How to add HID button
- [ ] How to add a different speech recognition library
- [ ] How to change sampling rate, bit resolution or number of input channels
- [ ] How to output raw mic signals)

keithm-xmos, 2023-04-12:
For "How to output raw mic signals"

xflash --factory example_ffva_verbose_output_test.xe  --boot-partition-size 0x100000 --data example_ffva_ua_adec_data_partition.bin

mbanth, 2023-09-15:
This issue is now tracked in [Carried Enhancements](https://xmosjira.atlassian.net/wiki/spaces/TNPMS/pages/3870916609/Carried+Enhancements).  The checklist in the description here should still be used to track pieces of work remaining to be completed.

closed 2023-09-15 by mbanth