xmos/sln_voice #275: Documentation updates for v2.0.0

keithm-xmos, 2023-05-08:
- [ ] Update programming guide to reflect audio pipelines that are now in modules
- [ ] Remove AVS setup instructions from Getting Started guide
- [ ] Update FFD documentation 
- [ ] Add low_power FFD documentation 
- [ ] Add doc (with working example) for changing/updating low-power FFD Sensory model
- [ ] Add entry about Sensory license to programming guide

keithm-xmos, 2023-06-07:
@jcarrier-xmos switched the style of two of the low-power FFD diagrams that are related to ones also in FFD. If the style he went with is preferred, the same update could be applied to FFD.

jcarrier-xmos, 2023-06-07:
Low power FFD does not have a `software_desc/asr.rst` (like ffd has, which is in need of updating) it is unclear how the `modules/asr` component is to be documented/linked to both FFD examples.

jcarrier-xmos, 2023-06-07:
I noticed that the `Third-Party` section in the docs needs to be updated for Sensory

closed 2023-06-23 by keithm-xmos