xmos/sln_voice #293: Does FFVA-UA work for 48KHz USB frequency

shuchitak, 2023-07-20:
If I change the appconfUSB_AUDIO_SAMPLE_RATE to 48000 in ffva.cmake and build and run the example_ffva_ua_adec_altarch.xe, I get very noisy output from the device. This doesn't happen all the time but frequently enough.
I've attached a couple of pipeline output wav files.
[ffva_48khz_pipeline_out_1.wav.zip](https://github.com/xmos/sln_voice/files/12107158/ffva_48khz_pipeline_out_1.wav.zip)
[ffva_48khz_pipeline_out.wav.zip](https://github.com/xmos/sln_voice/files/12107161/ffva_48khz_pipeline_out.wav.zip)

Also, if I enable debug prints in usb_audio.c, roughly every 16 seconds, I see one of the 2 messages that I've copied below. This happens every single time I record, starting from ~16 seconds into the recording.
**lost VFE output samples
Oops buffer is full**

Interestingly, if I disable the adaptive USB PLL nudging, I don't see the error prints at all.

mbanth, 2023-07-20:
@shuchitak, please provide a Story Point value for this issue.  Thanks.

mbanth, 2023-09-15:
This bug is now tracked in Jira [AP-120](https://xmosjira.atlassian.net/browse/AP-120).

closed 2023-09-15 by mbanth